BastilleResearch/gr-lora #6: Source - out(0): Port is not connected.

zoutepopcorn, 2016-11-16:
i'm a noob in using GNU radio.

I've some red blocks:
- pfb_arb_resampler
- 2x Missing block key lorademod

It said: Source - out(0):

Trying to connect an rtl2832 (dvb-t) stick to it. Any solutions?

Thanks for the cool work

matt-knight, 2016-12-31:
Thanks for your interest!  Please begin by verifying that you've installed GNU Radio and `gr-lora` on your system, and that your installation destinations are in your path.  `pfb_arb_resampler` is within the core GNU Radio package, and the LoRa package is included in `gr-lora`.  There are many guidelines on how to install GNU Radio for various OSes, including the official reference: http://gnuradio.org/redmine/projects/gnuradio/wiki/InstallingGR

matt-knight, 2017-01-30:
I'm going to mark this closed due to inactivity for now.  Feel free to re-open or create a new issue if you are still having trouble.  Thanks!

closed 2017-01-30 by matt-knight

zoutepopcorn, 2017-01-30:
Okido Thanks for your info