BastilleResearch/gr-lora #7: examples give me ```ImportError: No module named lora```

pointhi, 2016-12-29:
When I try to run the given gnuradio examples, they return:

```python
ImportError: No module named lora
```

The whole project was installed as described in the README.

matt-knight, 2016-12-31:
Two questions to get started:

1. Do you have GNU Radio installed and included in all relevant paths (executable, linking, includes, Python, etc.)?
2. Do you have `gr-lora` installed and included in said paths?

Forgive me if these questions are reductive, but installation and path issues are the most common source of your error.

herrfeuer, 2017-01-07:
Hi, i can reproduce this error on 2 Machines.
Debian Testing, gr and devel stuff out of apt repos. I am able to build other gr-modules on the same Installations without problems.
I tried to include some paths, but i had no success. What does need to be included ?

dc2mw, 2017-01-26:
Hello pointhi,

I have the same issue "No module named lora" but only on my Debian Jessie based old notebook. 

My newer Ubuntu 16.04 machines do not show this issue using the same instructions.
So a simple cause might be different default library path or environment variable setting ... as Matt said already.
I did not dig deeper into it as using another machine/distro was a quick solution for me.

matt-knight, 2017-01-27:
Thanks for your patience.  I was unable to replicate this on Ubuntu 14.04 or Debian 8.

Can I ask you to please post your distro+version, path contents (especially `$PYTHONPATH`), and install prefix (as indicated by `gr-lora/build/install_manifest.txt`)?  Feel free to anonymize or redact as appropriate.

mikaelnousiainen, 2017-01-27:
I had similar problems in Fedora 25 and I was able to solve the problem by adding `/usr/local/lib64/python2.7/site-packages` (or wherever the Makefile of gr-lora installs the packages) to a file ending with `.pth` extension in the standard `site-packages` directory `/usr/lib64/python2.7/site-packages` (which again may be different for your distribution).

So just create a file `/usr/lib64/python2.7/site-packages/usrlocal.pth` (the extension matters) which contains one line:
```
/usr/local/lib64/python2.7/site-packages
```

herrfeuer, 2017-01-27:
hi, my install_manifest contains:
/usr/local/lib/cmake/lora/loraConfig.cmake
/usr/local/include/lora/api.h
/usr/local/include/lora/demod.h
/usr/local/include/lora/decode.h
/usr/local/include/lora/mod.h
/usr/local/include/lora/encode.h
/usr/local/lib/libgnuradio-lora.so
/usr/local/lib/python2.7/dist-packages/lora/_lora_swig.so
/usr/local/lib/python2.7/dist-packages/lora/lora_swig.py
/usr/local/lib/python2.7/dist-packages/lora/lora_swig.pyc
/usr/local/lib/python2.7/dist-packages/lora/lora_swig.pyo
/usr/local/include/lora/lora/swig/lora_swig.i
/usr/local/include/lora/lora/swig/lora_swig_doc.i
/usr/local/lib/python2.7/dist-packages/lora/__init__.py
/usr/local/lib/python2.7/dist-packages/lora/__init__.pyc
/usr/local/lib/python2.7/dist-packages/lora/__init__.pyo
/usr/local/share/gnuradio/grc/blocks/lora_demod.xml
/usr/local/share/gnuradio/grc/blocks/lora_decode.xml
/usr/local/share/gnuradio/grc/blocks/lora_mod.xml
/usr/local/share/gnuradio/grc/blocks/lora_encode.xml
on Debian 8.6 x64.

I am sure the host is ok, i could build and run gr-ais, gr-ax25, gr-iee802-11, gr-dsd, and lots more
without any issues on the same Machine.

matt-knight, 2017-01-27:
@mikaelnousiainen Nice tip.  To root cause the initial issue, can you confirm whether `/usr/local/lib64/python2.7/site-packages` is in your `$PYTHONPATH` or `sys.path` when that `.pth` file is not in place?

@herrfeuer For your case, can you confirm whether `/usr/local/lib/python2.7/dist-packages` is in your `$PYTHONPATH` or `sys.path`?

matt-knight, 2018-02-23:
I'm going to close this due to inactivity -- feel free to re-open or start a new issue if you have further questions.

closed 2018-02-23 by matt-knight