BastilleResearch/gr-lora #8: Adding Encrypt/Decrypt LoRa

drtyhlpr, 2017-01-04:
inf thanks for this great research and thanks for your great talk! researching undocumented de-chirp, unwhite etc. may sound (to most people) far more easy then it is in RF land.

have you planned to include LoRA encrypt and decrypt functionality or would you be happy to a pull request with that :) ? (like https://github.com/jieter/python-lora).

matt-knight, 2017-01-27:
Hi @drtyhlpr, glad you enjoyed the talk!  I've had a great time working on this project and I'm so glad others are interested as well.

`gr-lora` is intended to stand as a dedicated physical layer.  MAC layer functionality is super interesting, but I think is best implemented in separate blocks.  This way `gr-lora` can remain protocol agnostic and work well with all MACs (LoRaMAC, LoRaHAM, etc).

I would be happy to advise or contribute to such a project.  As far as GNU Radio blocks go, the MAC parser you propose would be pretty easy to write since its I/O is through PDUs.  The API you linked to looks really straightforward, so in fact you might be able to wrap it within an `any_block` and skip writing a dedicated block for it: https://github.com/balint256/gr-baz/blob/master/grc/baz_any_code.xml

matt-knight, 2017-05-16:
Closing for inactivity -- feel free to re-open if you'd like to revisit the conversation.

closed 2017-05-16 by matt-knight