BastilleResearch/gr-lora #11: AttributeError when running examples/txrx_sim.grc

pgorczak, 2017-03-14:
When running examples/txrx_sim.grc, I get

```
File "/gr-lora/examples/top_block.py", line 89, in main
    tb = top_block_cls()
  File "/gr-lora/examples/top_block.py", line 35, in __init__
    self.lora_mod_0 = lora.mod(spreading_factor, 0x12)
AttributeError: 'module' object has no attribute 'mod'
```

Seems like unlike in #7, the module is found but it does not contain the `mod` function. Could this be an issue with swig? I'm running on Debian Stretch.

matt-knight, 2017-03-14:
Hi @pgorczak,

I'm not sure what the issue is, but I suspect it's related to #12.  Can you post your system and GNU Radio configuration so I can attempt to reproduce on my end?  OS distro and version, GNU Radio version/git hash, etc.

pgorczak, 2017-03-15:
Hi @matt-knight, thanks for your quick response!

I'm running a dockerized Debian GNU/Linux 9.0 (stretch) with GNU Radio 3.7.10 from the most recent version of master (0be096f1d724b27cd51582d341d7029e3ae038cd).

You could try my [Docker image](https://hub.docker.com/r/pgorczak/gnuradio/) if you want to reproduce it exactly. I put a [Dockerfile](https://gist.github.com/pgorczak/adca75e0938c087a2b2f845b0ced9916) up on Gist.

matt-knight, 2017-03-16:
@pgorczak Awesome, thanks for the detailed reply!  I'll try to reproduce over the next few days.  It's possible that with the GNURadio or Debian version bump some things got moved around.

pierreduf, 2017-07-19:
Hi,

I've got the same issue on Ubuntu 16.04 and GNU Radio v.3.7.9 installed from the official repos.

I know the thread is a bit old but have you finally managed to get it work ?

pierreduf, 2017-07-19:
I tried the "usrlocal.pth" trick in the #7 but without any success. My build manifest for the gr-lora module is as follows :

I have a lot of less file that in #7 . What could have happen ?

`/usr/local/lib/cmake/lora/loraConfig.cmake
/usr/local/include/lora/api.h
/usr/local/include/lora/demod.h
/usr/local/include/lora/decode.h
/usr/local/include/lora/mod.h
/usr/local/include/lora/encode.h
/usr/local/lib/libgnuradio-lora.so
/usr/local/lib/python2.7/dist-packages/lora/__init__.py
/usr/local/lib/python2.7/dist-packages/lora/__init__.pyc
/usr/local/lib/python2.7/dist-packages/lora/__init__.pyo
/usr/local/share/gnuradio/grc/blocks/lora_demod.xml
/usr/local/share/gnuradio/grc/blocks/lora_decode.xml
/usr/local/share/gnuradio/grc/blocks/lora_mod.xml
/usr/local/share/gnuradio/grc/blocks/lora_encode.xml`

pgorczak, 2017-07-19:
Hey @pierreduf, I'm sorry but I didn't get around to do more troubleshooting on this yet.

Flamewires, 2017-07-21:
try running <code>sudo ldconfig</code> after running make install.

matt-knight, 2017-07-21:
All,

I'm checking in to note that I haven't been able to reproduce this issue.  I just stood up the following in a VM without issues.

Ubuntu 16.04.1
GNU Radio 3.7.11
gr-lora d587169

For starters, please run `python -c "import sys; print '\n'.join(sys.path)"` and print the result here so we can verify your python path includes the install directories.  Share `echo $LD_LIBRARY_PATH` while you're at it too.  I always install to `/opt` and manually export paths so I'm curious if there's a discrepancy here.

If that's not fruitful, I'll ask you to share your system configuration and versions so I can attempt to reproduce.

Best,
Matt

matt-knight, 2017-07-24:
Given that several have had installation issues, I added some build and configuration guidelines to the Wiki: https://github.com/BastilleResearch/gr-lora/wiki/Installation

There are several viable ways to install GNU Radio and related modules, but this is the process I use.  If your method isn't working, document it here and I'll see if I can troubleshoot.

pgorczak, 2017-07-24:
I got around to trying again. When checking the CMake output, I discovered some suspicious output:

    [...]
    -- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
    -- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
    [...]
    -- Checking for module SWIG
    -- Disabling SWIG because version check failed.
    [...]

I guess no Doxygen is OK but I'll look further into the PkgConfig and SWIG issues.

pierreduf, 2017-07-24:
@Flamewires : I've made the ldconfig but made it again after your comment, unfortunately it doesn't change anything.

@matt-knight : this is the output of the command :

`pierred@pierred-vbox:~/gr-lora/build$ python -c "import sys; print '\n'.join(sys.path)"

/usr/lib/python2.7
/usr/lib/python2.7/plat-x86_64-linux-gnu
/usr/lib/python2.7/lib-tk
/usr/lib/python2.7/lib-old
/usr/lib/python2.7/lib-dynload
/usr/local/lib/python2.7/dist-packages
/usr/lib/python2.7/dist-packages
/usr/lib/python2.7/dist-packages/PILcompat
/usr/lib/python2.7/dist-packages/gtk-2.0
/usr/lib/python2.7/dist-packages/wx-3.0-gtk2
`

I think I'm gonna follow your updated installation guide : I guess something concerning the paths is messed up. I'll keep you updated.

pierreduf, 2017-07-24:
My bad ! After installing SWIG and making a clean install (following the new guide with the correct way to set path environment variables) everything is OK.

I think that missing SWIG was the problem, thanks @pgorczak for the hint !

No more errors now, the flowgraph is compiling and running ! I can't decode LoRa signals yet but that's another story.

Thank you again all for your support.

pgorczak, 2017-07-24:
Good news - I could get it to work on Ubuntu 16.04 :smile:

It must have been PkgConfig and/or SWIG which were missing but required for the Python build to work. Here is the new install output:

**These are the lines that were previously missing from the install output**
```
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/_lora_swig.so
-- Set runtime path of "/usr/local/lib/python2.7/dist-packages/lora/_lora_swig.so" to ""
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_swig.py
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_swig.pyc
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_swig.pyo
-- Installing: /usr/local/include/lora/lora/swig/lora_swig.i
-- Installing: /usr/local/include/lora/lora/swig/lora_swig_doc.i
```

I'll also give it a shot with Debian and update this comment.

closed 2017-07-24 by pgorczak

matt-knight, 2017-08-02:
Thanks @pgorczak and @pierreduf for working through this!