BastilleResearch/gr-lora #12: AttributeError: 'module' object has no attribute 'demod'

Nickes, 2017-03-14:
I'm not quite sure if this is really an issue, but I tried to use the lora-samples from rpp0 with this repository. I modified rx_ursp.grc to:
[image]
After executing the project, I'm getting the error message:
```
Executing: /usr/bin/python2 -u /home/user/gr-lora/examples/top_block.py

16:28:21: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1010,wx containers,compatible with 2.8),
and wxPython used 3.0 (wchar_t,compiler with C++ ABI 1009,wx containers,compatible with 2.8).
Traceback (most recent call last):
  File "/home/user/gr-lora/examples/top_block.py", line 236, in <module>
    main()
  File "/home/user/gr-lora/examples/top_block.py", line 230, in main
    tb = top_block_cls()
  File "/home/user/gr-lora/examples/top_block.py", line 133, in __init__
    self.lora_demod_0 = lora.demod(spreading_factor, ldr, 25.0, 2)
AttributeError: 'module' object has no attribute 'demod'
```
The path ```/usr/local/lib/python2.7/site-packages/lora/``` is in my ```$PYTHONPATH``` variable. 
Is it possible, to use this repo without a lora-device?

matt-knight, 2017-03-14:
Hi @nickes,

I think the issue is in fact with your `PYTHONPATH`.  Try chopping `lora/` off the end of the substring you listed.  So the string within `PYTHONPATH` would be `/usr/local/lib/python2.7/site-packages`.

Make sure you source/apply your path changes before testing (apologies if that's reductive).

matt-knight, 2017-03-14:
Actually you know what, I'm not sure that will do it.  Try it, but while you do can you post any build information I would need to reproduce?  i.e. GNU Radio version/commit hash, OS distort version, etc.

nickes, 2017-03-14:
Hi @matt-knight,
wow thanks for the quick reply. I'm using gnuradio version 3.7.10.1 on Arch Linux (4.9.11-1-ARCH, x86_64). I tried to shorten the path, but the result was the same.
I also noticed that I'm getting the same error like in the issue that you referenced.

matt-knight, 2017-03-16:
@nickes Sounds good, thanks for the details.  I'll try to reproduce and will update accordingly.

Wosser1sProductions, 2017-04-21:
@nickes For Arch Linux, you have to run `cmake` with the `-DCMAKE_INSTALL_PREFIX=/usr` argument.
(Same as in [rpp0/gr-lora](https://github.com/rpp0/gr-lora))

nickes, 2017-05-30:
@Wosser1sProductions I tried it with the argument, but the error is still the same. 
@matt-knight somehow I'm pretty sure that this is not really a problem of gr-lora, it has definitely something to do with my Arch environment.
I also tried to run the file on my Kali Linux distro on my laptop with gr 3.7.10. The flow graph executed without errors.
I think this issue can be closed...

closed 2017-05-30 by nickes

0xc0decafe, 2018-10-13:
Hi,

had the same problem when not installing to /opt prefix, the libgnuradio-lora.so was installed to /usr/lib instead of /usr/lib64. I guess it has to do with this part of the CMakeList.txt:

`set(GR_LIBRARY_DIR      lib${LIB_SUFFIX})
`

manually moving the file fixed the issue.

best