BastilleResearch/gr-lora #14: Gtk-Warning: attempt to allocate widget with width -5 and height 20.

vlad9319, 2017-05-10:
I run lora_receive_file.py for test decoding and getting an [Gtk-Warning: attempt to allocate widget with width -5 and height 20.
I tried decoding my lora and got infinitely symbol "O"(OOOOOOOOOOOO...).
 How to fix my problem?

matt-knight, 2017-05-16:
Hi @vlad9319,

Is `lora_receive_file.py` something you've written?  There is no such file in this repository.

The "O"s you are seeing are indicative of an overrun, or buffer overflow between your SDR and your PC.  Essentially your SDR is producing samples faster than your PC can consume them, so samples are being dropped as a result.  Try going after a narrower bandwidth signal, which will require a lower sampling rate, or by increasing the compute resources available to GNU Radio.

matt-knight, 2017-07-14:
I'm closing this due to inactivity -- feel free to create a new issue or reopen this one should you care to revisit it.

closed 2017-07-14 by matt-knight