BastilleResearch/gr-lora #15: gr-lora does not receive the messages

nickes, 2017-05-30:
I'm working with an HackRF One and a Dragino LoRA shield (v1.4, [this one](http://wiki.dragino.com/index.php?title=Lora_Shield)) + Arduino Uno. I also using the LMIC library with the example sketch ([Arduino sketch](https://github.com/matthijskooijman/arduino-lmic/tree/master/examples/raw)).
The setup is:
Frequency: 686MHz
Spreading factor: 8 (Can be set from 7 to 12)
BW: 125K 
The sketch sends a message ("Hello, world") every 5 seconds.
I tried several configurations of the rx_ursp sketch (I attached the modified file), but I can not receive any message.
I tried to receive the same message with gr-lora/rpp0 and it worked(also attached).
I'm not sure what I am doing wrong but I really would like to use this lora package to receive AND send lora packets.
Does somebody has the same experience or any clue, what the reason could be?
Both files are [here](https://1drv.ms/f/s!AnXjohrguxaRxE0omllrA0OzQmLJ)

RyabchikSib, 2017-06-03:
LoRa has 2 modes: explicit header (payload size in bytes, coding rate and header CRC(?)) and implicit (nothing before the payload). The whitening sequences for them are different, and only the last one is implemented in this project right know.

The receiver working in the explicit mode is implemented in the rpp0 project, but that sequence can't be used in this project for weird reasons (see Issue #13 for details)

nickes, 2017-06-04:
Thanks @RyabchikSib for your help,
those 2 modes sounds a bit confusing, I think I need to do some research...
My plan is to use my Arduino and gr-lora to capture and modify the messages.
Does anyone have some experiences with gr-lora and Arduino?

matt-knight, 2017-07-14:
Hi @nickes,

I've used the Adafruit LoRa modules throughout the course of development.  They're pretty nifty.

Demodulation will work regardless of the selected header mode -- the symbols will be correct, though the decoded values may not be.  If the demodulator is configured correctly and you're seeing nothing then there may be something else afoot.  I downloaded your sketch; if you can send an IQ file and its parameters I'll try to replicate on my end.

Best,
Matt

nickes, 2017-07-14:
hey @matt-knight,
thank you for reply. I have to correct my issue:  
At the moment I receive (some) messages, when the frequency of my dragino hat is set to `868.2 MHz`. The problem is, that the received messages are strangely modified. 
I'm preparing you a IQ file and more informationen as soon as possible. 
Are you using the Arduino IDE with the LMIC library to program your Adafruit LoRa module? 
     
BTW, I want to prepare a demo in which I'm using sdr (with my HackRF) to intercept encrypted LoRa messages. Afterwards I try to extract some important information from those messages to send corrupted ones (also through sdr). I hope this is somehow possible.

nickes, 2017-07-24:
Sorry for the late reply. I finally recorded the signal. 
The parameters are:

Frequency: 868.2MHz
Bandwitdh: 125K
Code Rate / Parity Bits: 4/8
Spreading factor: 8 
Implicit HDR: on
CRC: off
Message: "101010" or 31 30 31 30 31 30 
Interval: 3 sec

The signal [1] was recorded with the rotator and resampler, so it can be directly connected to the demodulator. The received signal is "31 30 31 10 34 38 e4".

[1] [https://1drv.ms/u/s!AnXjohrguxaRxE4_F_-HM97I7Q2Y](https://1drv.ms/u/s!AnXjohrguxaRxE4_F_-HM97I7Q2Y)

ThGeoffrey, 2018-02-05:
Hey @nickes, 
I know that it has been a long time since you've worked on that project, but I'm in the same position as you were in July. I'd like to make it work (I have a SX1272 + HackRF One trying to catch the former's Lora messages). 

I only managed to get @matt-knight 's gr-lora work for a SF8 BW250 CR8 CROFF HEADER OFF configuration. The other configurations do not work. And with rpp0's gr-lora, I manage to catch some caracters, but meaningless regarding to what I send... 

I'm quite desperate... Did you got something useful to help me please ? 

Thanks in advance...

PS : Sorry for my english if there are some mistakes... :dancer:

ThGeoffrey, 2018-02-09:
Hey @nickes ! 

Thanks for your answer ! I manage to have something with your help (I did not realize that I had to move a little bit the center frequency on the flowgraph in order to properly decode the message). 

Now, I have some results with both of the gr-LoRa versions : 
[image]

I'm sending the char "DEADBEEF" (44 45 41 44 42 45 45 46 in hex) and the picture shows what I receive with my hackRF One. 

Configuration : CH13 (866.1MHz), CR4/8, BW125, SF8, HOFF, CRCOFF

I guess I still have to work on the whitening sequence or the synchronization (I do not really know what the problem is : why I can decode the message for certain SF and not for all...). 

But thanks for your help !

nickes, 2018-02-10:
Hey, sounds good. 
I did exactly the same table-thing with the different spreading factors 😄.
When I played with the explicit header mode, I realised, that rpp0's gr-lora was quite stable with Header mode on (independent of the spreading factor).
Another advice: Don't forget that you can also play with the distance of the receiver and your transmitter. Somehow I received different messages, depending on the distance and the antenna I used.
Good luck!

bhomssi, 2019-07-06:
Hey everyone

Can anyone explain how this issue was fixed? I am facing the same exact problem and the message shows up on the FFT window however no data goes through the demod (I am using the example rx_usrp). I am using 921 MHz for my LoRa shield frequency (Tx) and 920 (MHz) for the RTL-SDR Source in gnuradio. I kept playing with the offset value in the slider but it hasn't been very generous either.

Thank you in advance