BastilleResearch/gr-lora #16: Resampling

RyabchikSib, 2017-06-03:
Hi!
I started working with SDR and GNU Radio not long ago and have a misunderstanding. 

As far as I understand, the generated chirps in /lib/mod_impl.cc have the bandwidth 2*pi.
Then in the flowgraph the decimation by factor sample_rate/bandwidth is applied and the signal is transmitted. What is the sampling frequency after the modulator? Is it the sampling rate, stated for the transmitter? How is this rate achieved? Does GNU Radio perform the interpolation itself?

The reason I'm asking is that I want to add noise after the modulator and to measure the error rate and can't figure out what sampling frequency to choose to get the right N0 value.

Each and every advice is appreciated)

RyabchikSib, 2017-06-12:
Can't get, what does it mean: "The width of the signal is determined by the sample rate."

matt-knight, 2017-07-14:
@RyabchikSib:

First, apologies for the delay!

This might be a bit of a tricky answer.  Within GNU Radio, there is no base sampling rate as I think you're alluding to.  Blocks must be synchronous (matched I/O rates and data types), and they publish/consume samples as fast as the flowgraph allows them to.  If there is a throttle or a radio sink/source within the flowgraph it will gate sample flow based on a clock reference, slowing everything down.  Without such an element, the flowgraph will run as fast as your CPU allows it to.

With regard to sample rate and bandwidth, the observable bandwidth is simply equal to the sample rate.  All that resampling is done to adjust the bandwidth of the signal.  The rest just works out based on how the LoRa physical layer is defined (the chirp rate being a function of the bandwidth and spreading factor).

If I've understood your goals correctly, I _think_ you'll want to introduce your voice source after the resampler -- you can do your math based on the signal bandwidth at that point.

Feel free to follow up if you have more questions and I promise I'll answer more promptly :)

matt-knight, 2018-02-23:
I'm going to close this due to inactivity -- feel free to re-open or start a new issue if you have further questions.

closed 2018-02-23 by matt-knight