BastilleResearch/gr-lora #17: which is  type of gr-lora input modulator?

stackprogramer, 2017-07-13:
it's input modulator has gray input, why we have not data type complex,float and...so  on........
your example can not work with file source in gnuradio
best regards stackprogramer

matt-knight, 2017-07-14:
Hi @stackprogramer,

The gray inputs you are seeing are for GNU Radio's messaging interface.  The easiest way to send/receive messages is to set up a `Socket PDU` within a flowgraph and then treat it like a web socket.  Try Linux's `nc` as I've done in the `README` to try it out.

Two advantages to the messaging interface are that it is asynchronous and really easy to interact with:

- A traditional streaming datatype (complex float) is used where the modulator and demodulator output/consume IQ, since those blocks touch the radio/DSP chain.  Asynchronous messaging enables the encoder and decoder to be bursty, since they only need to do work when there is a message ready to be sent or a packet has been received.
- Web sockets are super easy for passing data to other programs or a shell, or the `Message Debug` block if you want to keep it all in GRC.

What is in your file source?  What are you trying to do exactly?

Best,
Matt

matt-knight, 2018-02-23:
I'm going to close this due to inactivity -- feel free to re-open or start a new issue if you have further questions.

closed 2018-02-23 by matt-knight