BastilleResearch/gr-lora #20: Build succeeds even if SWIG is not found, breaking examples.

pgorczak, 2017-07-24:
*Just adding this for visibility. I'm going to look at the CMake files too.*

Building without SWIG seems to work fine but it breaks the examples, resulting e.g. in #11. Is SWIG always required for gr-lora to work? If it is, we should make the build fail if it can't be found.

matt-knight, 2017-08-02:
Hi @pgorczak,

Thanks for noting this.  While technically one can use GNU Radio without the Python layer (thus not requiring SWIG), I don't know a single person who does this.  So I am of the opinion that your first solution mandating the use of SWIG is the way to go.

I'm happy to handle this, unless you'd rather do the honors.

Best,
Matt

pgorczak, 2017-08-02:
Hey I just opened #21 after seeing how easy minor edits are from the website :)