BastilleResearch/gr-lora #25: gr-lora decode faild

cn0xroot, 2017-11-30:
Hi @matt-knight  I've been reading your ppt from
https://static1.squarespace.com/static/54cecce7e4b054df1848b5f9/t/57489e6e07eaa0105215dc6c/1464376943218/Reversing-Lora-Knight.pdf
It really cool！

But when i used gr-lora rx_usrp.grc to decode lora data ,nothing print on the nc console.

data samplefile：https://drive.google.com/file/d/1FniqHAiDO5owvK8XuVo5ErkYIP5PJ5QJ/view?usp=sharing

```
frequency：492.25e6
bandwith:125k
spreading_factor:11
code_rate:5
```

https://drive.google.com/open?id=1bmpmJA8X696Y5VAR5n06CxXcVwm0s0zQ

and the code_rate only 5-8(maybe) , it not 4?

Look forward to your reply,thanks!

matt-knight, 2018-02-23:
Hi @cn0xroot, sorry for the delay!  Glad you enjoyed the presentation.

How did you generate or capture your sample data file?  Let me know and I'll do some digging and try to reproduce on my end.

cn0xroot, 2018-02-23:
capture sample data file with rx_usrp.grc .   modify the frequency to 492.25e6 : https://drive.google.com/file/d/1bmpmJA8X696Y5VAR5n06CxXcVwm0s0zQ/view