BastilleResearch/gr-lora #27: tr_usrp.grc not transmitting via usrp

nilampradhan2, 2018-02-17:
i tried to transmit LoRa signal by running tr_usrp using USRP B200, but nothing is transmitted. red light LED is OFF only.

matt-knight, 2018-02-22:
Hi @nilampradhan2, two thoughts:

1) The flowgraph expects to be fed packets via the GNU Radio asynchronous message API.  The easiest way to do this is to use a UDP socket, as I have instrumented in the examples.  Run the flowgraph, connect to it via `$ nc`, type in some text, hit enter, and see what happens.

2) If this doesn't work, check your logs to ensure UHD isn't complaining.

Hope this helps,
Matt

nilampradhan2, 2018-02-23:
Thank you Matt for quick response.
Now it is working with Random socket PDU, but when i select socket PDU
source and doing same procedure as you have mentioned, it unable to
transmit. Red LED is OFF only.
I will try it again.

Regards,

Nilam Pradhan
Asst. Prof.
E & TC Dept.
MITCOE, Pune

nilampradhan2, 2018-03-01:
> Sounds like it's a network configuration issue. Try UDP mode, and nc over
> localhost and the appropriate port.

Thanks for reply.
My transmitter is working with udp socket PDU.
RED LED turns ON for few seconds. After sending over it shows letter U in
log and receiver socket pdu don't receive anything. I give nc command and
port address.

brunoricardojava, 2018-03-28:
@nilampradhan2 could you tell which modifications made it work?

nilampradhan2, 2018-03-29:
I set parameters as frequency 868Mhz, code rate 4, sampling rate 250 khz,
bw 250 khz , sf 8.
Header false, transmit gain 50

hningning, 2018-08-03:
Hi,@nilampradhan2  Have you solved this question?I opened two terminals and connected to the flowgraph via
$ nc -l -u localhost 52002   and  $ nc -u localhost 52001. After sending over it shows letter U in log just the same as you mentioned before.