BastilleResearch/gr-lora #39: Decode data from RN2483

DmitriiCM, 2019-06-06:
Hi,

Thanks for the cool product.
Interesting thing to test, but I was not able to get proper data from RN2483.

I sending this data to the modem:
```
mac pause
radio set mod lora
radio set sf sf8
radio set freq 869100000
radio set cr 4/8
radio tx 1234
```
With 'rpp0/gr-lora' I can decode the data properly, and get output like '02 90 70 12 34 c8 ea'  - "12 34" is there.

With BastilleResearch/gr-lora I am getting wrong data: 'df f7 2b 56 5b 02'.
Parameters (SF, bw, others looks the same).

Is it a bug, or I'm doing something wrong or this library is not compatible with RN2483? 
GRC and IQ files in attach.
[lora_test.zip](https://github.com/BastilleResearch/gr-lora/files/3260717/lora_test.zip)

Thanks.

rokigeorg, 2019-06-06:
Hey,

to be honest, i cant really help you on this.
Its been to long for me that I have used the RN2483.

Forget about the rpp0/gr-lora. It might be buggie.
Only use BastilleResearch/gr-lora its the original repo.

Also I remember, when I used tried to establish a Peer-to-Peer connction between to Lora modules (HopeRF Lora module) I got some junk data before the payload.
It looked something likes this:
--------------------------------
| Junk1 | Junk2 | Junk3|Payload|
--------------------------------

I never found the reason for it.
Maybe Preamble stuff ...
Maybe is the LoraWan Header ...

Back then, i decided to cut of the junk of the received Package.
I had a deadline :-)

Let me know if you find out the reason for the junk data in the header.

riatankarsahu, 2020-09-02:
Hello @DmitriiCM ,
I am now a days working with gr-lora and also having the same problem. I am using LimeSDR to receive the signal. Did you figure out the solution? 
I will be highly glad to have them if you have any..... Thanks in advance!!!
-- Ritankar