BastilleResearch/gr-lora #41: Issue with Decoding!!

riatankarsahu, 2020-07-28:
Hello Matt,

Hope you are doing fine due to this pandemic. I was trying to do my Bachelor's Engineering Thesis on LoRa communication with GnuRadio and I found your project gr-lora really useful for it. But somehow I am experiencing some problem with Demodulation and Decoding blocks.

I am using LimeSDR (Chipset - LMS7002M) as the hardware, and gnuradio version 3.7. While demodulating and decoding, I am receiving different values than transmitted values, but I have seen in your video on youtube, that you could receive everything perfectly.

So I am now getting surprised about my setup and all, if I made some mistakes or not. I am using the LoRa standard of Europe now as I am from Poland. Hence, if you please guide me a little bit through it, I will be really grateful to you. Thank you in advance!! Smile

With Warm Regards

--
Ritankar Sahu

Student of B.Sc [Macro]

DEPARTMENT OF AUTOMATICS, ELECTRONICS AND COMPUTER SCIENCE
SILESIAN UNIVERSITY OF TECHNOLOGY

ph03n1xAu, 2020-11-23:
Hi Ritankar @riatankarsahu,

Hope you found some solutions using LimeSDR for LoRa decode. I am also trying to do the same with LimeSDR but failed to decode. Not sure though whether it is due to any transmission issues or only the decoding problem. 

Just wondering, could you share some updates or how you resolved the issues

Thanks..

riatankarsahu, 2020-11-23:
Hello phoenix(@ph03n1xAu ),

I couldn't find any solution yet, but I think that repository is written for particular hardware set. I saw in presentation of Matt that he could decode the message but I couldn't with different method and hardware. I even have some simulation *iq file which was wrongly decoded by this block. 
I also saw that it can demodulate the signal in any aspect, but decoding is quite faulty. According to the hardware, B210 and LimeSDR are not very different by Sample rate and Sample depth
[image]
you can see in the picture. Matt used B210, so LimeSDR should work in same way. I think it's about the transmitter.  I am after it and I hope I can find some solution. If I find something I will definitely inform you.
Good Luck!
-- Ritam

ph03n1xAu, 2021-04-11:
Hi Ritam @riatankarsahu ,

Hope your studies going well. Just wondering have you found any fix for Lora Transmitter with LimeSDR? I'm trying to make this work with Lime but no outcome yet. I'd love to collaborate if you are still working on this.

Cheers!!

riatankarsahu, 2021-04-12:
Hey @ph03n1xAu ,
How are you doing? I was working with the Receiver actually. So not yet I have something for the transmission part. This very repository is working with 'Implicit Header' mode. I think for 'Explicit Header' I will have to write myself something. How about your receiver? Hope it's working well!

ph03n1xAu, 2021-04-13:
Yeah, rppo/receiver is working fine with LimeSDR with Lora hardware transmitter. But I need to use SDR as transmitter. I'm working with both this transmitter library and rppo/encoder library.. but both requires modification of the actual code. 

Initially, we can go with implicit header to test limesdr as transmitter. Unfortunately, I could not able to use this repo to work with limesdr, couldn't figure out the actual cause yet.

riatankarsahu, 2021-04-13:
I am actually working nowadays with RPi integration with SDR. Give me some times. I will check the transmission section myself after a few days. You can contact me by mail if you wish!

ph03n1xAu, 2021-04-14:
Oh, it would be great if you can share your email address :-)
I did few experiments on embedded platform like pi's with LimeSDR. Let me know if you need help