BastilleResearch/gr-lora #46: Camke error

Mkishore1, 2023-07-13:
hello i am getting the error for CMP00XX ...
can anyone help me to resolve this error.

-- Build type not specified: defaulting to release.
CMake Deprecation Warning at CMakeLists.txt:39 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:44 (cmake_policy):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.35") found components: filesystem system 
-- Using GMP.
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.71.0") found components: date_time program_options system regex thread unit_test_framework 
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake:35 (if):
  if given arguments:

    "ON"

  An argument named "ON" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  CMakeLists.txt:124 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake:
  Policy CMP0011 is not set: Included scripts do automatic cmake_policy PUSH
  and POP.  Run "cmake --help-policy CMP0011" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  The included script

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake

  affects policy settings.  CMake is implying the NO_POLICY_SCOPE option for
  compatibility, so the effects are applied to the including context.
Call Stack (most recent call first):
  CMakeLists.txt:124 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:128 (message):
  GnuRadio Runtime required to compile lora

-- Configuring incomplete, errors occurred!
See also "/home/vboxuser/gr-lora/build/CMakeFiles/CMakeOutput.log".
See also "/home/vboxuser/gr-lora/build/CMakeFiles/CMakeError.log".

heyadidadida, 2024-01-19:
I have encountered the same problem. Have you resolved it？

Tomson96, 2024-10-07:
Does anybody was able to resolve this issue?

eabase, 2024-11-06:
maybe try this one: 
https://github.com/rpp0/gr-lora